jneilliii/OctoPrint-TPLinkSmartplug #361: [BUG]:Printer does not turn off after the idle period

Anathema-Device, 2024-03-17:
**Describe the bug**
Configuring the printer to be turned off after a 30 minute idle time does not seem to result in it being turned off.  The printer can be turned off and on with the icon that the plugin puts in the Octoprint toolbar but does not turn off when the printer is idle 

**To Reproduce**
Steps to reproduce the behavior:
1. Enable the checkbox to turn off on idle
2. Set the idle delay to be 30 minutes
3. Print something
4. Wait for it to turn off and nothing happens.

**Expected behavior**
The TPLink smart plug should be turned off after the idle time

**Desktop (please complete the following information):**
 - Windows 11
 - Brave
 - OctoPrint Version : current release
 - Plugin Version : current version
 
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/14628644/plugin_tplinksmartplug_debug.log)

lethuer, 2024-04-01:
I also observe that described behaviour. My idle time is set to 3min...
Any recommendation ?
Also M81 in terminal does not turn off the plug.
Lightning button is working as expected.

jneilliii, 2024-04-01:
Both of you need to share screenshots of your settings please. For M81 to work you have to include the IP address of the plug you want to turn off.

lethuer, 2024-04-01:
Here are my settings.
[image]
[image]

For M81 I included the ip address... 
But the plug didn't switched off.

jneilliii, 2024-04-01:
@lethuer for gcode to work you have to enable the gcode trigger option.

lethuer, 2024-04-01:
Awesome ! That works immediately in terminal :) 

Now let me ask: If gcode trigger is enabled, is M81 command send automatically and under which conditions ?
Is it possible to send M81 command when a desired nozzle temp is reached after a print ?

jneilliii, 2024-04-01:
You could do that in your end gcode of your slicer or in OctoPrint's gcode script section. Can't remember if sending the wait for temp command will block until the temp is reached and then allow power off.

lethuer, 2024-04-01:
Okay I will try if it is waited until reaching desired nozzle temperature...

Is here the correct location ?
And is also the ip address needed or just M81 required ?
[image]

lethuer, 2024-04-01:
I also configured M81 code in section of cancelling a print.
Unfortunately I can't observe in terminal that M81 is sent.
Not directly after aborting the print and also not when nozzle is reaching 50 degree again.
Maybe because printer has changes it's status to "Operational" before reaching 50 degree ?

`Send: N16 M118 P0 A1 action:cancel*14
Recv: //action:cancel
Cancelling on request of the printer...
Recv: ok N16 P15 B3
Send: N17 M84*41
Recv: ok N17 P15 B3
Send: N18 M104 T0 S0*24
Recv: ok N18 P15 B3
Send: N19 M140 S0*93
Recv: //action:notification Bed Cooling...
Recv: ok N19 P15 B3
Send: N20 M106 S0*85
Recv: M106 P0 S0
Recv: ok N20 P15 B3
Changing monitoring state from "Cancelling" to "Operational"
[...]
Send: M117 L=-/- E=-
Recv: ok P15 B3
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"`

lethuer, 2024-04-03:
I tried again and now received this:

`Changing monitoring state from "Printing" to "Cancelling"
Send: N15 M108*30
Recv: ok N14 P15 B3
Recv: ok N15 P15 B3
Send: N16 M117 L=-/83 E=-*62
Recv: ok N16 P15 B3
Send: N17 M118 P0 A1 action:cancel*15
Recv: //action:cancel
Cancelling on request of the printer...
Recv: ok N17 P15 B3
Send: N18 M84*38
Recv: ok N18 P15 B3
Send: N19 M104 T0 S0*25
Recv: ok N19 P15 B3
Send: N20 M140 S0*87
Recv: //action:notification Bed Cooling...
Recv: ok N20 P15 B3
Send: N21 M106 S0*84
Recv: M106 P0 S0
Recv: ok N21 P15 B3
**Send: N22 M81 192.168.178.46*29**
Recv: M106 P0 S0
Recv: //action:notification Artillery Genius (lethuer) OFF.
Recv: ok N22 P15 B3`

I think the problem was that in my previous post I used @TPLINKOFF 192.168.178.46 instead of M81 192.168.178.46 in gcode section.

So now M81 was sent which I can observe in the terminal.
But nethertheless the plug didn't turned off.
Maybe because temperature was higher than needed for idle ?
If I manually type M81 in the terminal the plug switches of immediately, so my assumption is that the plugin does prevent this in gcode section ?

jneilliii, 2024-04-04:
Based on your printer response to m81 I suspect the other option of using @TPLINKOFF command would be better. Also, typically what I see with gcode processing is that the delay isn't long enough to allow the final commands from the file to clear the printing buffer. If you want me to assist further I need you to enable debug logging in the plugin's settings and restart OctoPrint. After attempting and it fails share plugin_tplinksmartplug_debug.log. That will help determine what the plugin is doing.

lethuer, 2024-04-04:
> Based on your printer response to m81 I suspect the other option of using @TPLINKOFF command would be better

But I didn't observe any command is sent if @TPLINKOFF was set up in gcode script section

> If you want me to assist further I need you to enable debug logging in the plugin's settings and restart OctoPrint. After attempting and it fails share plugin_tplinksmartplug_debug.log. That will help determine what the plugin is doing.

Logging was active the hole time, here is the file also containing yesterday evening. Can you work with that ? 
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/14862820/plugin_tplinksmartplug_debug.log)

1st try with @TPLINKOFF
2nd try with M81

lethuer, 2024-04-04:
And the new file with repeated test:

M81 configured in gcode script section
09:28 plug switches off (idle)
09:29 turn on
09:29 start print
09:30 cancel print (nozzle temp in heating phase about 70 degree)
09:36 plug switches off (idle)

@TPLINKOFF configured in gcode script section
09:37 turn on
09:37 start print
09:38 cancel print (nozzle temp in heating phase about 80 degree)
09:43 plug switches off (idle)

Terminal output from last attempt:
```
Changing monitoring state from "Printing" to "Cancelling"
Send: N14 M108*31
Recv: ok N13 P15 B2
Recv: ok N14 P15 B3
Send: N15 M117 L=-/124 E=-*1
Recv: ok N15 P15 B3
Send: N16 M118 P0 A1 action:cancel*14
Recv: //action:cancel
Cancelling on request of the printer...
Recv: ok N16 P15 B3
Send: N17 M84*41
Recv: ok N17 P15 B3
Send: N18 M104 T0 S0*24
Recv: ok N18 P15 B3
Send: N19 M140 S0*93
Recv: //action:notification Bed Cooling...
Recv: ok N19 P15 B3
Send: N20 M106 S0*85
Recv: M106 P0 S0
Recv: ok N20 P15 B3
Changing monitoring state from "Cancelling" to "Operational"
[...]
Send: M117 L=-/- E=-
Recv: ok P15 B3
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"
```

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/14863108/plugin_tplinksmartplug_debug.log)

lethuer, 2024-04-04:
I observed there weren't any [DEBUG] info in the last logfile (don't know why not this time but before...)
Anyhow again I repeatet the test:

**M81 configured in gcode script section**
09:57 turn on
09:57 start print
09:58 cancel print (nozzle temp in heating phase about 90 degree)
10:03 plug switches off (idle)

terminal output:
```
Recv: echo:busy: processing
[...]
Changing monitoring state from "Printing" to "Cancelling"
Send: N16 M108*29
Recv: ok N15 P15 B3
Recv: ok N16 P15 B3
Send: N17 M117 L=-/124 E=10:22*21
Recv: ok N17 P15 B3
Send: N18 M117 L=-/124 E=-*12
Recv: ok N18 P15 B3
Send: N19 M118 P0 A1 action:cancel*1
Recv: //action:cancel
Cancelling on request of the printer...
Recv: ok N19 P15 B3
Send: N20 M84*45
Recv: ok N20 P15 B3
Send: N21 M104 T0 S0*18
Recv: ok N21 P15 B3
Send: N22 M140 S0*85
Recv: //action:notification Bed Cooling...
Recv: ok N22 P15 B3
Send: N23 M106 S0*86
Recv: M106 P0 S0
Recv: ok N23 P15 B3
Send: N24 M81 192.168.178.46*27
Recv: M106 P0 S0
Recv: //action:notification Artillery Genius (lethuer) OFF.
Recv: ok N24 P15 B3
Changing monitoring state from "Cancelling" to "Operational"
[...]
Send: M117 L=-/- E=-
Recv: ok P15 B3
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"

**@TPLINKOFF configured in gcode script section**
10:05 turn on
10:05 start print
10:07 cancel print (nozzle temp in heating phase about 70 degree)
10:11 plug switches off (idle)

terminal output:
```
Changing monitoring state from "Printing" to "Cancelling"
Send: N14 M108*31
Recv: ok N13 P15 B3
Recv: ok N14 P15 B3
Send: N15 M117 L=-/124 E=-*1
Recv: ok N15 P15 B3
Send: N16 M118 P0 A1 action:cancel*14
Recv: //action:cancel
Cancelling on request of the printer...
Recv: ok N16 P15 B3
Send: N17 M84*41
Recv: ok N17 P15 B3
Send: N18 M104 T0 S0*24
Recv: ok N18 P15 B3
Send: N19 M140 S0*93
Recv: //action:notification Bed Cooling...
Recv: ok N19 P15 B3
Send: N20 M106 S0*85
Recv: M106 P0 S0
Recv: ok N20 P15 B3
Changing monitoring state from "Cancelling" to "Operational"
[...]
Send: M117 L=-/- E=-
Recv: ok P15 B3
[...]
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"
```

Again I didn't observe any command is sent if @TPLINKOFF was set up in gcode script section

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/14863442/plugin_tplinksmartplug_debug.log)

jneilliii, 2024-04-04:
ok, so idle timeout is working properly. honestly that's the preferred approach as it will account for temperatures to be below a certain threshold before powering off. 

It's normal to not see @ commands in the terminal, because those aren't sent to the printer, they are specifically used internally by OctoPrint. It is strange that it doesn't work as expected. Based on your comment you added that command to OctoPrint's on cancel gcode script?

lethuer, 2024-04-04:
Correct, just like this.
[image]

jneilliii, 2024-04-04:
Thanks I'll do some additional testing with gcode scripts being used to trigger. There's a chance that those commands might not go through the processing queue like the file commands do that the plugin is using to do the work.

lethuer, 2024-11-08:
Hi ! I again observe this issue that printer does not turn off after printing...
Did you have the chance to look into it ?

jneilliii, 2024-11-08:
Sorry, no, my focus has been on one of the many other plugins I maintain lately. Did you still have debug logging enabled? Could you share an updated log file if so. Also, by chance are you running the current stable version or the latest rc version?

lethuer, 2024-11-08:
Here is an updated log file, print finished at 21:15
[20241108_plugin_tplinksmartplug_debug.log](https://github.com/user-attachments/files/17683930/20241108_plugin_tplinksmartplug_debug.log)

Settings:
[image]

[image]

jneilliii, 2024-11-09:
Your settings are probably preventing the process from happening properly. Turn off run gcode command, and increase the GCODE off delay to allow the printer's printing buffer to complete and the printing status to change away from "printing". I'd start with trying 30 seconds. 

[image]

lethuer, 2024-11-09:
GCode was disabled before but I never tried with the delay.
Now I did but this makes no difference.

[image]

jneilliii, 2024-11-09:
Did you happen to restart OctoPrint after the change?

lethuer, 2024-11-09:
Yes sure :)

jneilliii, 2024-11-09:
ok, had a chance to look at the actual log and noticed this...

> resetting idle timer due to ClientOpened event.

but that was way after the idle timeout, any errors in octoprint.log?

what version are you running of the plugin? I just tested on my OctoPrint instance and powered on the printer from the button, it auto-connected and after the 3 minute timeout I got the message in the UI to allow me to cancel the power off if I wanted to and then it powered off. 

[image]

[image]

lethuer, 2024-11-11:
> what version are you running of the plugin?

Version: 1.0.3
[image]

> I just tested on my OctoPrint instance and powered on the printer from the button, it auto-connected and after the 3 minute timeout I got the message in the UI to allow me to cancel the power off if I wanted to and then it powered off.

Yes. this is also working for me but the printer does not turn off after a print.

This were the settings of the print today:
[image]

And the tp link log file:
[20241111_plugin_tplinksmartplug_debug.log](https://github.com/user-attachments/files/17707543/20241111_plugin_tplinksmartplug_debug.log)

Print was started at 16:33 and finished at 19:48.
At start I have this message:
`[2024-11-11 16:33:25,096] DEBUG: Resetting idle timer since plug 192.168.178.46 was just turned on.`
and approx 40min after finishing this one:
`[2024-11-11 20:29:46,344] DEBUG: resetting idle timer due to ClientOpened event.`
no such messages during printing...

Also I have enabled this one for logging:
[image]

I'm not really sure what to search for. The file is to large to upload here, but here is a download link:
`https://1drv.ms/u/s!AmTrnLTKnK36iuoUUfBW8yk-SdGiKw?e=DGPUPO`

jneilliii, 2024-11-11:
let's see if the issue persists in the current rc version. in OctoPrint's software update settings change the release channel for the plugin to `release candidate` and update when prompted.

lethuer, 2024-11-12:
Ok. Did so and started a print...
We'll see in a few hours :)

[image]

lethuer, 2024-11-12:
Okay print was started at 19:19 and finished at 22:34, nozzle temp of about 50 degree (idle target temperature) was reached again at 22:45.
TP Link plug did not turn off after print, here is the log with messaged until 23:14
[20241112_plugin_tplinksmartplug_debug_1.0.4rc4.log](https://github.com/user-attachments/files/17723368/20241112_plugin_tplinksmartplug_debug_1.0.4rc4.log)

I found this with "idle timer"

```
[2024-11-12 19:05:50,271] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-12 19:08:20,011] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-12 19:09:16,622] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-12 19:11:12,073] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-12 19:12:25,945] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-12 19:13:15,389] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-12 19:14:36,816] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-12 19:14:56,599] DEBUG: Resetting idle timer since plug 192.168.178.46 was just turned on.
[2024-11-12 19:19:17,964] DEBUG: Resetting idle timer since plug 192.168.178.46 was just turned on.
[2024-11-12 23:14:09,227] DEBUG: resetting idle timer due to ClientOpened event.
```

jneilliii, 2024-11-12:
ok, this ClientOpened event apparently is getting in the way here. I don't know what application/plugin that is causing that, but it might be the telegram plugin? Basically the idle timer is getting reset because it detects "someone" connecting to OctoPrint's web interface, or potentially directly to the server/API backend. 

Instead of using idle timers it seems you may need to use gcode triggers to get around that. Change your settings to this:
[image]

and then in OctoPrint's gcode scripts for after print completes and cancelled sections add the line `@TPLINKOFF 192.168.178.46`
[image]

lethuer, 2024-11-13:
> ok, this ClientOpened event apparently is getting in the way here. I don't know what application/plugin that is causing that, but it might be the telegram plugin?

I tried to disable telegram to check if this is the problem. TP link again did not switch off after the print was done.
Print was started at 11:00 and finished at 14:15, here is the log from that
[20241113_plugin_tplinksmartplug_debug_telegram_disabled.log](https://github.com/user-attachments/files/17734436/20241113_plugin_tplinksmartplug_debug_telegram_disabled.log)

I can still observe this ClientOpened event (so telegram seems not to be the issue): 
How to evaluate which addin is responsible for those ClientOpened events ? 
```
[2024-11-13 10:54:54,694] DEBUG: Settings saved, Automatic Power Off Enabled, starting idle timer...
[2024-11-13 10:55:49,293] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-13 10:59:34,259] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-13 10:59:39,056] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-13 10:59:44,693] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-13 10:59:54,730] DEBUG: Resetting idle timer since plug 192.168.178.46 was just turned on.
[2024-11-13 15:56:10,817] DEBUG: resetting idle timer due to ClientOpened event.
```

> Instead of using idle timers it seems you may need to use gcode triggers to get around that. 

Also tried this, here tp link does immediately switch off after the print was done.
It takes about 10 minutes to reach 50 degree again so would you recommend to set delay of gcode off to 10*60sec = 600 sec ?
[image]

jneilliii, 2024-11-13:
> It takes about 10 minutes to reach 50 degree again so would you recommend to set delay of gcode off to 10*60sec = 600 sec ?

Yes, since the gcode triggers don't support cooldown you can either increase this delay out to an estimated time frame, or alternatively use a cooldown command in your slicer's end gcode setting to do that for you. Adding the below line to the slicer's end gcode would wait until temp is reached.

```
M109 R50
```

this way the OctoPrint gcode scripts for print completion would not trigger immediately because the print job wouldn't complete until that 50 was reached. unfortunately this wouldn't work for cancelled prints so the long delay may be the way to go. 

I don't know if there is a way to determine what client is being opened to troubleshoot that, but it's either someone opening the webpage or an external service/plugin doing it.

lethuer, 2024-11-13:
This is a list of my installed plugins.
Any ideas which one could be the problem ?

[image]

[image]

[image]

jneilliii, 2024-11-14:
if you enabled debug logging for octoprint.events.fire, it might help you determine where that is coming from.

lethuer, 2024-11-15:
> if you enabled debug logging for octoprint.events.fire, it might help you determine where that is coming from.

[plugin_tplinksmartplug_debug (2).log](https://github.com/user-attachments/files/17774222/plugin_tplinksmartplug_debug.2.log)

Print started at 08:55 and finished at 10:19.
In that log file for example I can find:
```
[2024-11-15 08:53:31,400] DEBUG: Settings saved, Automatic Power Off Enabled, starting idle timer...
[2024-11-15 08:54:55,108] DEBUG: Resetting idle timer since plug 192.168.178.46 was just turned on.
[2024-11-15 08:56:51,280] DEBUG: resetting idle timer due to ClientOpened event.
[2024-11-15 08:56:57,412] DEBUG: resetting idle timer due to ClientOpened event.

```

After I enabled debug logging for octoprint.events.fire in octoprint.log I can see these "ClientOpened" events:
```
2024-11-15 08:56:51,268 - octoprint.events.fire - DEBUG - Firing event: ClientOpened (Payload: {'remoteAddress': '192.168.178.159'})
2024-11-15 08:56:57,396 - octoprint.events.fire - DEBUG - Firing event: ClientOpened (Payload: {'remoteAddress': '192.168.178.159'})
2024-11-15 11:28:29,880 - octoprint.events.fire - DEBUG - Firing event: ClientOpened (Payload: {'remoteAddress': '192.168.178.159'})
```
This is my laptop, maybe because I had the web view opened (but I have to to start the print job... )
No additional "ClientOpened" events beside these and tp link didn't turn off.

lethuer, 2024-11-15:
> > ~ahh..ssd oled display might be the one.~ maybe not looking at its code.
> 
> I give it a try and will disable the plugin.

Okay disabled the plugin and I'm wondering why there are still messages updated on the ssd display.
Anyhow, tp link does not switch off after the print when ssd add in is disabled.

jneilliii, 2024-11-15:
I'm at a loss then for your issue. I am unable to reproduce and everything works fine for me with idle timeout.

lethuer, 2024-11-22:
I think I've found the plugin which causes the problem...
Not 100% sure right now but if I disable "DisplayLayerProgress Plugin" tp link switches off after a print and reaching 50deg again.

I investigated a bit and it seems that as soon as I disable "Drucker Anzeige" in plugin settings of "DisplayLayerProgress Plugin" tp link does switch off after a print and reaching target temp again.
[image]

jneilliii, 2024-11-22:
Not sure why that would impact my plugin, that's just sending M117 commands to the printer to display on printer screen. You might be able to add that to tplink's ignored commands and it would allow for the idle to be detected.

lethuer, 2024-11-22:
I reproduced the observed bahavior and created a log file with "Drucker Anzeige" set on where tp link didn't switch off after print.
Print started at 15:10 and finished at 15:48.
File is to large to upload here and zip is forbidden. You can download it here: https://1drv.ms/u/s!AmTrnLTKnK36iuxHlbkJyexTaQH4GA?e=1E423h

Here is the tp link log from that aswell
[20241122 plugin_tplinksmartplug_debug.log](https://github.com/user-attachments/files/17873285/20241122.plugin_tplinksmartplug_debug.log)

lethuer, 2024-11-22:
Ignoring M117 commands does not help

lethuer, 2024-11-22:
Also ignoring M118 commands does not help.

jneilliii, 2024-11-23:
It definitely looks like that plugin is sending over and over the last layer of layer and current time to the printer with M117. Did you add the M117 like this? 

```
M105,M117
```

[image]

lethuer, 2024-11-23:
Yep, I tried exactly like that.
So commaseparated list

lethuer, 2024-11-23:
Just for Information: I activated ssd1306 plugin again which also should send M117 messages. With that active tp link does also switch off after print.

jneilliii, 2024-11-23:
Also, you entered it exactly like that, with no spaces between any characters?

lethuer, 2024-11-24:
With no spaces it is working :)
[image]

[image]

lethuer, 2024-11-24:
Again I made a test with a space in between, again it does not turn off.
Here is a log from that, print startet at 15:16 and finished at 19:36
[20241124 serial.log](https://github.com/user-attachments/files/17894583/20241124.serial.log)

Last M117 message of regular printing:
`2024-11-24 19:35:42,306 - Send: N142416 M117 L=96/96 E=19:36*1`
From here on once in a minute M117 message is sent:
```
2024-11-24 19:37:00,960 - Send: M117 L=96/96 E=19:37
2024-11-24 19:38:01,030 - Send: M117 L=96/96 E=19:38
2024-11-24 19:39:01,101 - Send: M117 L=96/96 E=19:39
2024-11-24 19:40:01,172 - Send: M117 L=96/96 E=19:40
```

So I will do following things:
- add M117 to ignore list of tp link without comma
- deactivate "Drucker Anzeige" of plugin "DisplayLayerProgress"
- activate ssd1306 plugin again or remove additional external oled display

jneilliii, 2024-11-25:
Yes, this makes sense to me. The whole input field is split by comma only, not come and space. I have an idea on how to fix that, but am away on vacation right now.

jneilliii, 2024-11-26:
Just released 1.0.4rc5 with a fix for the space issue.

lethuer, 2024-12-06:
> Just released 1.0.4rc5 with a fix for the space issue.

tested and works for me !

jneilliii, 2024-12-06:
@Anathema-Device did you happen to also have the DisplayLayerProgress plugin installed in OctoPrint?

lethuer, 2024-12-07:
Did you really mean @Anathema-Device ?

I have DisplayLayerProgress installed but without "Drucker Anzeige" and now your fix for spaces in field for ignored g code commands just seem to work :)

jneilliii, 2024-12-07:
> Did you really mean @Anathema-Device ?

Yes, they're the one that originally posted this issue.

Anathema-Device, 2024-12-07:
> @Anathema-Device did you happen to also have the DisplayLayerProgress plugin installed in OctoPrint?

Not specifically, I did see that one plugin was writing messages to the terminal after printing had finished and I just nuked that one, so now I just have the basic set of stuff, Marlin docs and plugins specific to the printer.  I have not been relying on the TP Link plugin to shut things down and try and walk over and push the button on the smartplug to turn it on or off.

This should be a list of the plugins that I currently have and the terminal is quiet now when the print is done but I will admit that I have not checked out the log to see if there was any messages saying why it does not turn off automatically 15 mins after the print is finished.

Plugin Licenses
BGCode: AGPLv3
Cancel Objects: AGPLv3
Cura Thumbnails: AGPLv3
Dashboard: AGPLv3
DisplayZ: AGPLv3
Exclude Region: AGPLv3
FileManager: AGPLv3
Marlin GCode Documentation: AGPLv3
OctoApp: AGPLv3
Prusa Filament Runout: AGPLv3
Segmented Bed: AGPLv3
Slicer Thumbnails: AGPLv3
Spoolman: AGPLv3
TP-Link Smartplug: AGPLv3
UI Customizer: AGPLv3

BTW, sorry about the slow response I did do the email reply thing but it seems that my email is not sending for some reason, just another issue to figure out, grrr ...

jneilliii, 2024-12-07:
Just took a second look at your debug log @Anathema-Device and your issue I think is related to this line...

> [2024-03-17 00:53:18,109] DEBUG: Turning off heater: X5

I believe you would also benefit from the latest RC version and adding X5 to the added ignore heaters list. Similar to what's described in #336

Anathema-Device, 2024-12-07:
OK, thanks, I will get the rc version and see if it helps.  I notice that Octoprint seems to find 8 heaters (0-5 plus X and the bed).  Since the UI only shows heaters 0-4 should I also add 5 to the exclude list or is 5 (tool5) and X5 the same thing?

[2024-12-07 08:48:12,429] DEBUG: Idle timeout reached after 10 minute(s). Turning heaters off prior to powering off plugs.
[2024-12-07 08:48:12,430] DEBUG: Heater tool0 already off.
[2024-12-07 08:48:12,431] DEBUG: Heater tool1 already off.
[2024-12-07 08:48:12,431] DEBUG: Heater tool2 already off.
[2024-12-07 08:48:12,431] DEBUG: Heater tool3 already off.
[2024-12-07 08:48:12,432] DEBUG: Heater tool4 already off.
[2024-12-07 08:48:12,432] DEBUG: Heater tool5 already off.
[2024-12-07 08:48:12,432] DEBUG: Heater bed already off.
[2024-12-07 08:48:12,433] DEBUG: Turning off heater: X5

jneilliii, 2024-12-07:
Do you have 5 heads? If so your printer profile settings in OctoPrint should be set that way to show all the heaters in the UI. Since it's detecting that tool5 as already off I don't think it's necessary to also add it. Do you have a XL printer? Might need to add X1,X2,X3,X4,X5 to the ignore list.

Anathema-Device, 2024-12-07:
I got the new code and put X,X5,tool5 in the heater exclude list and it seemed to power off from just sitting and not having been run.  With the improved logging (thanks!) I see that there is a whole raft of other "heaters" so I should probably add A and all the B_* heaters to the exclude list but it seems to turn off now.

2024-12-07 14:23:56,399] DEBUG: Idle timeout reached after 10 minute(s). Turning heaters off prior to powering off plugs.
[2024-12-07 14:23:56,401] DEBUG: Heater tool0 already off.
[2024-12-07 14:23:56,402] DEBUG: Heater tool1 already off.
[2024-12-07 14:23:56,402] DEBUG: Heater tool2 already off.
[2024-12-07 14:23:56,402] DEBUG: Heater tool3 already off.
[2024-12-07 14:23:56,403] DEBUG: Heater tool4 already off.
[2024-12-07 14:23:56,403] DEBUG: Heater bed already off.
[2024-12-07 14:23:56,403] DEBUG: Heater A already off.
[2024-12-07 14:23:56,403] DEBUG: Heater B_0_0 already off.
[2024-12-07 14:23:56,404] DEBUG: Heater B_1_0 already off.
[2024-12-07 14:23:56,404] DEBUG: Heater B_2_0 already off.
[2024-12-07 14:23:56,404] DEBUG: Heater B_3_0 already off.
[2024-12-07 14:23:56,404] DEBUG: Heater B_0_1 already off.
[2024-12-07 14:23:56,405] DEBUG: Heater B_1_1 already off.
[2024-12-07 14:23:56,405] DEBUG: Heater B_2_1 already off.
[2024-12-07 14:23:56,405] DEBUG: Heater B_3_1 already off.
[2024-12-07 14:23:56,406] DEBUG: Heater B_0_2 already off.
[2024-12-07 14:23:56,406] DEBUG: Heater B_1_2 already off.
[2024-12-07 14:23:56,406] DEBUG: Heater B_2_2 already off.
[2024-12-07 14:23:56,406] DEBUG: Heater B_3_2 already off.
[2024-12-07 14:23:56,407] DEBUG: Heater B_0_3 already off.
[2024-12-07 14:23:56,407] DEBUG: Heater B_1_3 already off.
[2024-12-07 14:23:56,407] DEBUG: Heater B_2_3 already off.
[2024-12-07 14:23:56,407] DEBUG: Heater B_3_3 already off.
[2024-12-07 14:23:56,408] DEBUG: Heater tool0 = 22.0C
[2024-12-07 14:23:56,408] DEBUG: Heater tool1 = 24.0C
[2024-12-07 14:23:56,408] DEBUG: Heater tool2 = 22.0C
[2024-12-07 14:23:56,409] DEBUG: Heater tool3 = 22.0C
[2024-12-07 14:23:56,409] DEBUG: Heater tool4 = 22.0C
[2024-12-07 14:23:56,409] DEBUG: Heaters below temperature.
[2024-12-07 14:23:56,409] DEBUG: Checking for timelapse running.
[2024-12-07 14:23:56,410] DEBUG: Checking timelapse status before shutting off power...
[2024-12-07 14:23:56,410] DEBUG: Starting abort power off timer.
[2024-12-07 14:24:26,481] DEBUG: Automatically powering off enabled plugs.

jneilliii, 2024-12-07:
You do want to leave the B ones, those are actually the different bed segments on your printer and you want those to cool down before petting off I suppose.

Anathema-Device, 2024-12-08:
OK, I left the B_* but added the A heater to the ignore list and it seemed to work OK.  That is, if I turn the machine on then it will turn off after the idle time, yay!  Then I deleted the log to start a clean one and ran a real print.  The print completed just fine but the machine was still on this morning, sigh.  Is there something that I need to put in the end Gcode to say that the job is finished and to power down?

[6 more replies]