jneilliii/OctoPrint-TPLinkSmartplug #362: [BUG]:  OctoPrint Toolbar toggle does not work with TPLink KP400, BUT @TPLINKON/OFF commands DO work

Dweinbach, 2024-04-11:
**Describe the bug**
OctoPrint Toolbar toggle does not work with TPLink KP400, BUT @TPLINKON/OFF commands DO work

**To Reproduce**
Steps to reproduce the behavior:
1. set up IP address of TPLink KP400
2. send @TPLINKON 192.168.xxx.xxx then @TPLINKOFF 192.168.xxx.xxx  in terminal. light will go on and off.
3. Click the lightning bolt in status bar of octoprint - no response.
4. See debug log

**Expected behavior**
If @ON and @OFF work, I would expect toolbar to, even if this is an unsupported device.

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: macos, windows11
 - Browser chrome
 - OctoPrint Version [latest]
 - Plugin Version [latest]
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/14949578/plugin_tplinksmartplug_debug.log)

**Additional context**
Add any other context about the problem here.

jneilliii, 2024-04-11:
That's weird. The logic for turning on and off is the same no matter which method you use, it calls the same function. Can you confirm if the KP400 in the Kasa app can control individual sockets separately or not? If it does, then it should work the same as my HS300, and using `192.168.0.196/1` (first socket) for IP address would maybe help.

jneilliii, 2024-04-11:
If that doesn't help I'll need you to open your browser's developer tools (F12) and switch to the network tab, click the button and see what the response is from the request.

Dweinbach, 2024-04-11:
> That's weird. The logic for turning on and off is the same no matter which method you use, it calls the same function. Can you confirm if the KP400 in the Kasa app can control individual sockets [...]

Thanks. That helped. Here is what I figured out:
IN TERMINAL, @ commands can be specified as 192.168.xxx.xxx, 192.168.xxx.xxx/1, or 192.168.xxx.xxx/2. if I leave off the /, it toggles BOTH channels of the device. That is what I was doing before.
IN TOOLBAR, I set up all three versions of the IP address listed above, as separate devices. It turns out the versions WITH / both work and operate the correct channel, but unlike in terminal, the version without the / does nothing. not sure why. Tried it on other browsers with same result.

jneilliii, 2024-04-11:
Cool, so with the `/` everything is working as intended it seems.

closed 2024-04-11 by jneilliii