jneilliii/OctoPrint-TPLinkSmartplug #364: [BUG]: Lightning turning never ending

lethuer, 2025-05-12:
**Describe the bug**
Plug turns on / off when clicking the lightning indicator but turns and keeps red without stopping.

**Expected behavior**
Should stop turning once it is connected and then should become grren.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome, others untested
 - OctoPrint Version 1.11.0
 - Plugin Version 1.0.4rc6 (1.0.4rc5 aswell)

**Additional context**
Add any other context about the problem here.

jneilliii, 2025-05-17:
please enable debug logging in the plugin settings, restart octoprint and then re-create the issue. then share the tplinksmartplug_debug.log that will be seen in OctoPrint's logging section of settings.

jneilliii, 2025-05-21:
ok, nothing out of the ordinary there in the log and it seems that your plug is reporting being off, so the red indicator on the lightning bolt is correct. can you now please open your browser's developer tools window while on the OctoPrint web interface and press the refresh button and look for any errors on the console tab.

lethuer, 2025-05-21:
didn't switched on after the reboot...
so plug was always off in log I think.

`Failed to load resource: the server responded with a status of 500 (INTERNAL SERVER ERROR)`

[image]

jneilliii, 2025-05-21:
ok, internal server error will probably have something indicated in octoprint.log why it's happening. can you please share that file with me?

jneilliii, 2025-05-22:
Perfect, thank you. That does help a lot.
```
2025-05-22 20:33:51,413 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
Traceback (most recent call last):
  File "/home/lethuer/oprint/lib/python3.9/site-packages/octoprint/server/api/__init__.py", line 93, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/lethuer/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1683, in wrapper
    return f(*args, **kwargs)
  File "/home/lethuer/oprint/lib/python3.9/site-packages/octoprint_tplinksmartplug/__init__.py", line 629, in on_api_get
    response = self.check_status(request.args.get("checkStatus"))
  File "/home/lethuer/oprint/lib/python3.9/site-packages/octoprint_tplinksmartplug/__init__.py", line 577, in check_status
    if self.last_row_entered is False and last_p == 0 and p > 0: #Go back & enter last_row on power return (if not entered already)
AttributeError: 'tplinksmartplugPlugin' object has no attribute 'last_row_entered'
```
This is the underlying issue that is causing the status check to not complete, and is related to the database logging of energy consumption. I'll take a deeper look into the code on this and get back with you.

jneilliii, 2025-05-22:
actually, I think I have this issue already resolved in the latest RC release. Can you go into OctoPrint's Software Update settings, find the plugin in the list and change its release channel to `Release Candidate` and upgrade when prompted. After that is installed, please confirm if it's working or not for you then.

jneilliii, 2025-05-22:
Sorry, I misspoke. This looks like one I may have missed, expect a 1.0.4rc7 version that will resolve this error hopefully.

lethuer, 2025-05-22:
Thanks for the quick analysis.
I'm running on the last release candidate 1.0.4rc6, I will try 1.0.4rc7 as soon it is available.
If you like you can give me a short hint here when it's online :)

[image]

jneilliii, 2025-05-22:
It's up now, bit you might have to force check in software update settings to bypass the daily cache, or it will show once that cache expires.

jneilliii, 2025-05-22:
For reference, to force check...

[image]

lethuer, 2025-05-24:
Forced today the search for updates but didn't show up ?

[image]

jneilliii, 2025-05-24:
Try now.

lethuer, 2025-05-25:
Updated to 1.0.4rc7 and no turning lightning anymore.
So looks like it's working now :)

commit 2ad8f601 (2025-05-22): 1.0.4rc7

add last_row_entered to init to hopefully resolve #364
closed 2025-08-13 by jneilliii