jneilliii/OctoPrint-TPLinkSmartplug #365: [Feature Request]: Tapo Device Support

Hatsune-Cthulhu, 2025-07-23:
I know you've discussed not supporting Tapo in the past, but given that Kasa has been discontinued for some time now, the only devices that can now be purchased are Tapo. Is there any chance you'd reconsider this stance?

If not, this plugin is going to become unusable for new users that are unable to attain the devices that this plugin supports unless they happen to be lucky enough to already have them lying around.

jneilliii, 2025-07-23:
Luckily for you I have recently seen wider support with some well maintained Python modules that I would consider adding to the plugin to work with those. Unfortunately, it's going to more than likely require a full refactor/rewrite of the plugin to integrate, which will require some time to research and implement.

slash1div, 2025-07-23:
Lol, feeling really dumb right now, I bought a Tapo plug for $15 without reading close enough since I could only find the broken Kasa plugs at my local Walmart. Entirely my fault. I would love to see Tapo support in the future, and I will be holding onto mine for the other 3D printer I have in the case it does get added.

jneilliii, 2025-08-17:
Version 2.0.0rc1 is out and is working with my Kasa based HS300, Kasa based HS103 and Tapo based P115. You can change the release channel in OctoPrint's Software Update settings to Release Candidate and update when prompted to get this version. 

Things I've noted so far, the P115 doesn't provide as much energy data as the kasa devices do. Tapo devices will require entering your username and password registered in the Tapo app, Kasa devices don't require those settings to be entered.

jneilliii, 2025-08-17:
I've confirmed the new refactor also supports Smart lights. When I bought the pair of P115 plugs I also got a pair of Tapo L530E bulbs and on/off works a treat. Will be migrating the features of my unlisted [OctoPrint-TPLinkSmartBulb](https://github.com/jneilliii/OctoPrint-TPLinkSmartBulb) plugin over to this one to allow for controlling color/brightness, etc.

jneilliii, 2025-08-17:
That wasn't too difficult to integrate, latest 2.0.0rc2 version now supports [M150 ](https://reprap.org/wiki/G-code#M150:_Set_LED_color) and [M355](https://reprap.org/wiki/G-code#M355:_Turn_case_lights_on.2Foff) gcode commands. Just have to enable the option in the device's individual settings.

[image]

sgiraz, 2026-10-17:
Test report for **2.0.0rc7 with a Tapo P300** (3-socket power strip), in case it helps getting 2.0 out of RC.

OctoPrint 1.11.8 · Python 3.13.16 · Raspberry Pi 3B+ (aarch64, Docker) · python-kasa 0.11.0.1 · P300 hw 1.0 / fw 1.4.2 (KLAP v2)

Works: on/off of a single strip socket from the UI, automatic idle power-off (serial disconnect first, then off), and no credentials in the logs.

Found along the way:
- #378 — autoConnect never fires for strip sockets: `is_on` is stale right after switching, because python-kasa refreshes a child from its parent's update. Fix in #380, tested on the P300.
- #379 — `ip/N` follows the firmware's child order, not the physical socket order: on the P300 the printer on socket 1 is `/3`.
- #381 — removes the `idle timer exception: 'NoneType' object has no attribute 'is_alive'` error logged on every start.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

jneilliii, 2026-10-17:
> KLAP v2

Nice, thanks for reporting this is working for you with python-kasa 0.11.0.1. That means I can pin this version in the plugin and resolve #371 as well.

jneilliii, 2026-10-18:
New version 2.0.0 is out with Tapo support.

closed 2026-10-18 by jneilliii