jneilliii/OctoPrint-TPLinkSmartplug #369: [BUG]: "On with Upload" no longer works with PrusaSlicer's "Upload and Print"

timur-tabi, 2025-08-24:
**Describe the bug**
I don't know what changed, but for the past week, when I click on "Upload and Print" in PrusaSlicer, the TP-Link plug-in no longer automatically turns on the printer.  I have to manually turn it on, wait 20 seconds for TP-Link to tell Octoprint to connect, and then manually start the print from Octoprint.  Previously, I used to be able to avoid connecting to Octoprint completely to print from PrusaSlicer.

I didn't change any settings.  "On with Upload" is enabled with the printer, as is Auto Connect (20 seconds).

OctoPrint 1.11.2
Python 3.9.2
OctoPi* 1.0.0cam (build 2025.04.22.120926)

timur-tabi, 2025-08-24:
The log shows this:

[2025-08-24 12:23:11,243] INFO: Turning off 192.168.1.22/3 at 2025-08-24 12:23:11.243121
[2025-08-24 12:23:12,120] INFO: Turning off 192.168.1.22/2 at 2025-08-24 12:23:12.120323
[2025-08-24 12:23:14,015] INFO: Turning off 192.168.1.39 at 2025-08-24 12:23:14.014906

I'm not sure why it's turning off my devices when it should be turning them on.  192.168.1.22/2 is the printer.

jneilliii, 2025-08-24:
Interesting, if it comes back feel free to re-open this issue.